dart-archive/sdk #488: Dartino Atom plugin should configure file tree view to exclude build artifacts

mit-mit, 2016-05-10:
After a few flashes the tree view on the right showing files becomes quite cluttered. Would be great if it could be configured to exclude build artifacts, incl.:

*.snapshot
*.bin
*.info.json

sgjesse, 2016-05-10:
We should also put all build artifacts into a build directory.

The question is how to structure this. E.g.
```
my-project/dartino.yaml
my-project/test1/test.dart
my-project/test2/test.dart
```
Now running (in my-project):
```
dartino build test1/test.dart
dartino build test2/test.dart
```
Should probably create the following
```
my-project/build/test1/test.bin
my-project/build/test1/test.elf
my-project/build/test1/test.snapshot
my-project/build/test1/test.info.json
my-project/build/test2/test.bin
my-project/build/test2/test.elf
my-project/build/test2/test.snapshot
my-project/build/test2/test.info.json
```

mit-mit, 2016-05-19:
@danrubel what do you think?

danrubel, 2016-05-19:
I like the idea. Let me see what can be done.

mit-mit, 2016-05-30:
Tracking the build part in https://github.com/dartino/sdk/issues/513

danrubel, 2016-05-31:
After digging into this further, this would be a fair amount of work and be counter to the approach of most other Atom plugins. I recommend [putting all build artifacts into a build directory](https://github.com/dartino/sdk/issues/513) and not filter the tree view.

mit-mit, 2016-06-01:
I'm OK with that suggestion @danrubel. Closing this.

closed 2016-06-01 by mit-mit