dart-archive/sdk #495: Strange extra information when trying to compile a non existent file

ricowind, 2016-05-12:
How to reproduce
```
$ dartino run /superman/is/dead
Using template settings file '/usr/local/google/home/ricow/temp/codelab/dartino-sdk/internal/.dartino-settings'
/superman/is/dead:
Error: Error reading '/superman/is/dead' (No such file or directory)
Compile error: main is malformed.

null
```
The last 3 lines here does not really add anything but confusion

ricowind, 2016-05-12:
Also, when running flashify on a non existing file it goes south:
Error: Error reading 'samples/general/hello.dart' (No such file or directory)
Cannot open file '/usr/local/google/home/ricow/temp/codelab/samples/general/hello.snapshot' for writing.
No such file or directory.
FileSystemException: Cannot open file, path = '/usr/local/google/home/ricow/temp/codelab/samples/general/hello.snapshot.info.json' (OS Error: No such file or directory, errno = 2)
#0      _File.open.<anonymous closure> (dart:io/file_impl.dart:356)
#1      _rootRunUnary (dart:async/zone.dart:902)
#2      _CustomZone.runUnary (dart:async/zone.dart:798)
#3      _Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:551)
#4      _Future._propagateToListeners (dart:async/future_impl.dart:637)
#5      _Future._completeWithValue (dart:async/future_impl.dart:424)
#6      _Future._asyncComplete.<anonymous closure> (dart:async/future_impl.dart:479)
#7      _rootRun (dart:async/zone.dart:895)
#8      _CustomZone.run (dart:async/zone.dart:790)
#9      _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:723)
#10     _microtaskLoop (dart:async/schedule_microtask.dart:41)
#11     _startMicrotaskLoop (dart:async/schedule_microtask.dart:50)
#12     _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#13     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:149)

Required third party tools GCC ARM Embedded and OpenOCD have not been installed.

Try running x-download-tools to add these tools to the Dartino install.

sigurdm, 2016-05-13:
Proposed fix: https://codereview.chromium.org/1975103002/

closed 2016-05-17 by sigurdm