dart-archive/sdk #496: Port Dartino to MIPS32 platforms

petar-jovanovic, 2016-05-13:
Given the popularity, presence and energy efficiency of MIPS CPUs in embedded systems, we believe support for MIPS CPUs in Dartino is of great importance. For this reason, RT-RK has done a port of Dartino for MIPS, and we would like to discuss it with you, get your feedback on the code and changes, and hopefully merge it into the trunk after reviews.

You can find [the initial port](https://github.com/petar-jovanovic/sdk/tree/mips-dev) at the link below and [instructions ](https://github.com/petar-jovanovic/sdk/wiki/Getting-started-with-MIPS-Dartino) for trying it out at one of the wiki pages.

Looking forward to hearing back from you on this!

Refs.
[1] MIPS on Dartino, code, https://github.com/petar-jovanovic/sdk/tree/mips-dev
[2] Getting started with MIPS Dartino, https://github.com/petar-jovanovic/sdk/wiki/Getting-started-with-MIPS-Dartino

sgjesse, 2016-05-17:
Thanks a lot for working on this!

I will try it out shortly, but I don't see any blockers for landing MIPS support in the tree.

For a change of this size, we would appreciate, if you can upload the patch(es) using `git cl upload` for reviewing in Rietveld (https://codereview.chromium.org/) for the actual code review.

From your Git repository the change is already split into 5 different patches, which might work well for landing as well.

petar-jovanovic, 2016-05-25:
First patch uploaded at
https://codereview.chromium.org/2006403003/

petar-jovanovic, 2016-06-02:
Second patch uploaded at
https://codereview.chromium.org/2029943003/

petar-jovanovic, 2016-09-02:
This issue can be closed as fixed. Since the last change was committed 23 days ago, all tests pass for QEMU MIPS32.

[45:05 | 100% | + 5283 | -    0]

closed 2016-09-02 by petar-jovanovic