dart-archive/sdk #499: When no network is available the fallback configuration does not take effect

sgjesse, 2016-05-18:
The following code keeps looping waiting for `NetworkInterface.list()` to become non-empty.

Shouldn't the fallback configuration take effect in that case?
```
// Initialize the network stack and wait until the network interface has either
// received an IP address using DHCP or given up and used the provided
// fallback [address].
const fallbackAddress = const InternetAddress(const <int>[192, 168, 0, 10]);
const fallbackNetmask = const InternetAddress(const <int>[255, 255, 255, 0]);
const fallbackGateway = const InternetAddress(const <int>[192, 168, 0, 1]);
const fallbackDnsServer = const InternetAddress(const <int>[8, 8, 8, 8]);

void initializeNetwork({
  InternetAddress address: fallbackAddress,
  InternetAddress netmask: fallbackNetmask,
  InternetAddress gateway: fallbackGateway,
  InternetAddress dnsServer: fallbackDnsServer}) {

  if (!ethernet.InitializeNetworkStack(address, netmask, gateway, dnsServer)) {
    throw "Failed to initialize network stack";
  }

  while (NetworkInterface.list().isEmpty) {
    print("Waiting for network interface to come up");
    sleep(100);
  }

  NetworkInterface iface = (NetworkInterface.list())[0];

  print("Network setup done ${iface.addresses[0]}");
}
```

sgjesse, 2016-05-19:
After an offline discussion I checked this once again. The issue with `NetworkInterface.list()` being empty only happens when the ethernet port is not connected at all. When the port is connected the network interface comes up. And it has the fallback configuration if there is no DHCP server present.

karlklose, 2016-06-01:
The library has been changed to always include the network interface (3c7e311e91a4874fac2de997602bca522398877c) and https://github.com/dartino/sdk/issues/521 tracks the issue with static configuration vs. DHCP.

closed 2016-06-01 by karlklose