dart-archive/sdk #504: Debugging fails on a mac

mit-mit, 2016-05-19:
When attaching, it just hangs trying to connect.

sgjesse, 2016-05-19:
This is must likely caused by the serial port configuration not being correct. Right now we just use `dart:io `to open the device (e.g. `/dev/ttyACM0`), and `dart:io` does not give any control over the serial port settings.

Right now the device is configured for 115200,8,n,1

And on Linux this matches the default:

```
$ stty -F /dev/ttyACM0 
speed 115200 baud; line = 0;
kill = ^H; eof = ^A; min = 1; time = 0;
-icrnl -imaxbel
-opost -onlcr
-isig -icanon -iexten -echo -echoe -echok -echoctl -echoke
```

sgjesse, 2016-05-19:
But on Mac the default is:

```
$ stty -f /dev/tty.usbmodem1413 
speed 9600 baud;
lflags: -icanon -isig -iexten -echo
iflags: -icrnl -ixon -ixany -imaxbel -brkint
oflags: -opost -onlcr -oxtabs
cflags: cs8 -parenb crtscts
```

We probably need a native extension to handle this correctly.

sigurdm, 2016-05-20:
This page (http://www.clearchain.com/blog/posts/using-serial-devices-in-freebsd-how-to-set-a-terminal-baud-rate)
 explains the problem somewhat. It seems that on BSD (and therefore probably also on macOS) changes to the tty settings are reset when the program setting them stops. Therefore stty cannot change it. But the page explains that there are tty.init files that can be used for permanent settings. 
I'm also looking into using the serial_port package with its native extension.

sigurdm, 2016-06-06:
The serial port should work on a mac as of https://github.com/dartino/sdk/commit/33fde1ba34b38f32188ce682d1abb6bc95511597 . I am still waiting for equipment, can anyone with a mac confirm that it works?.

mit-mit, 2016-06-07:
I'd be happy to test this. Can you update the instructions [here](https://github.com/dartino/sdk/wiki/Debugging-via-serial-on-the-STM-Discovery-board), please?

sigurdm, 2016-06-07:
Yes - good idea! Done.

mit-mit, 2016-06-07:
It works great!

closed 2016-06-07 by mit-mit