dart-archive/sdk #509: Analytics messages either consist of content or startup/shutdown messages

lukechurch, 2016-05-22:
Messages to analytics either look like:

Where '____' is one or more fields that are omitted for privacy or brevity reasons.

```
~1463685115617:startup:______:0.4.0-edge._____:1.15.0-edge._____ (Wed Feb 17 18::46::10 2016) on "linux_x64":linux
~1463685115664:shutdown
```

or like this:

```
~1463685115707:request:______ :debug:serve: ______
```

They don't contain both a startup/shutdown message and content as to what was happening. This means that it's not possible to look at anything beyond global usage patterns.

@danrubel @mit-mit

danrubel, 2016-05-23:
Looks like we have analytics reporting startup/shutdown in the main isolate and reporting request information in the worker isolate. I'll investigate.

lukechurch, 2016-05-23:
Thanks @danrubel

danrubel, 2016-05-26:
Should now be fixed in head. Please reopen if not.

closed 2016-05-26 by danrubel