dart-archive/sdk #510: Lines sample can crash on STM32F746G Discovery

sgjesse, 2016-05-23:
The sample runs for some time - typically between 10 and 60 seconds, and then crashes.

Here is one sample crash from a debug build:

```
(gdb) bt
#0  0x0804b62e in _exit (status=<optimized out>)
    at ../../platforms/stm/disco_dartino/src/syscalls.c:57
#1  0x0807eb72 in abort ()
#2  0x08010ffa in dartino::Platform::ImmediateAbort ()
    at ../../src/shared/platform_cmsis.cc:134
#3  0x0801158a in dartino::DynamicAssertionHelper::Fail<dartino::DynamicAssertio
nHelper::ASSERT> (file=<optimized out>, line=<optimized out>,
    format=0x8097678 "expected: %s") at ../../src/shared/assert.cc:37
#4  0x08033f2c in dartino::CompactingVisitor::Visit (this=0x20017bb4,
    object=0xc012e825) at ../../src/vm/object_memory_mark_sweep.cc:419
#5  0x0803667a in dartino::Space::IterateObjects (this=this@entry=0x20024c50,
    visitor=visitor@entry=0x20017bb4) at ../../src/vm/object_memory.cc:151
#6  0x0802e918 in dartino::Program::CompactSharedHeap (
    this=this@entry=0x20024ae8) at ../../src/vm/program.cc:595
#7  0x0802eb28 in dartino::Program::PerformSharedGarbageCollection (
    this=this@entry=0x20024ae8) at ../../src/vm/program.cc:535
#8  0x0802eb9c in dartino::Program::CollectOldSpace (
    this=this@entry=0x20024ae8) at ../../src/vm/program.cc:504
#9  0x0802ec0a in dartino::Program::CollectOldSpaceIfNeeded (
    this=this@entry=0x20024ae8) at ../../src/vm/program.cc:1139
#10 0x0802f2e6 in dartino::Program::CollectNewSpace (this=0x20024ae8)
    at ../../src/vm/program.cc:1131
#11 0x08012232 in dartino::HandleGC (process=0x20024ea0)
    at ../../src/vm/interpreter.cc:123
#12 0x08000338 in InterpreterEntry ()
    at obj/src/vm/dartino_vm_interpreter_library.gen/generated.S:74
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) 
```

Other types of crashes has been observed. This is on 4f792c625f28dce92ebcf52ba0a8007f129593d3.

sgjesse, 2016-05-23:
Here is a different stacktrace:

```
(gdb) bt
#0  0x0804b62e in _exit (status=<optimized out>)
    at ../../platforms/stm/disco_dartino/src/syscalls.c:57
#1  0x0807eb72 in abort ()
#2  0x08010ffa in dartino::Platform::ImmediateAbort ()
    at ../../src/shared/platform_cmsis.cc:134
#3  0x0801158a in dartino::DynamicAssertionHelper::Fail<dartino::DynamicAssertionHelper::ASSERT> (file=<optimized out>, line=<optimized out>, 
    format=0x8097678 "expected: %s") at ../../src/shared/assert.cc:37
#4  0x08038eb0 in dartino::Array::cast (object=0x0)
    at ../../src/vm/object.h:1867
#5  dartino::Class::methods (
    this=0x2000651d <dartino::StaticClassStructures::free_list_chunk_class_storage+1>) at ../../src/vm/object.h:1974
#6  dartino::Class::LookupMethod (
    this=this@entry=0x2000651d <dartino::StaticClassStructures::free_list_chunk_class_storage+1>, selector=selector@entry=91392) at ../../src/vm/object.cc:470
#7  0x08012dee in dartino::HandleEnterNoSuchMethod (process=0x20024ea0)
    at ../../src/vm/interpreter.cc:325
#8  0x080018b4 in BC_EnterNoSuchMethod ()
    at obj/src/vm/dartino_vm_interpreter_library.gen/generated.S:3121
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) 
```

sgjesse, 2016-05-23:
And a crash in the interpreter:

```
(gdb) bt
#0  0x0804b67a in hardfault (frame=0x20017f00, regs=0x2004ffc0)
    at ../../platforms/stm/disco_dartino/src/exceptions.c:180
#1  <signal handler called>
#2  0x08087234 in O3325c234 ()
#3  0xfffffc00 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) p/x *frame
$1 = {r0 = 0x0, r1 = 0x808f571, r2 = 0x55555555, r3 = 0x3ca, r12 = 0x8086001,
  lr = 0x80009a5, pc = 0x8000980, psr = 0x1000000}
(gdb) list *$1->lr
0x80009a5 is at obj/src/vm/dartino_vm_interpreter_library.gen/generated.S:1027.
(gdb) list *$1->pc
0x8000980 is at obj/src/vm/dartino_vm_interpreter_library.gen/generated.S:1014.
(gdb) p/x *regs
$2 = {r4 = 0x20024ea0, r5 = 0x808be0b, r6 = 0xc01ab434, r7 = 0x644,
  r8 = 0x8086001, r9 = 0xfffffc00, r10 = 0x8087235, r11 = 0x8087181}
```

ErikCorryGoogle, 2016-05-30:
This issue was fixed by removing __builtin_prefetch from the GC marker. Apparently the code was too highly optimized.

closed 2016-05-30 by ErikCorryGoogle