dart-archive/sdk #512: Dartino Atom analysis fails for Dart Atom users

mit-mit, 2016-05-27:
The analysis in the Dartino Atom plugin fails if a user has installed regular Dart before installing Dartino:

## Repro steps

On a clean Ubuntu machine:

1. Install dart:
```
sudo sh -c 'curl https://storage.googleapis.com/download.dartlang.org/linux/debian/dart_stable.list > /etc/apt/sources.list.d/dart_stable.list'
sudo apt-get update
sudo apt-get install dart
```

2. Install atom
```
sudo apt-get install git
wget https://github.com/atom/atom/releases/download/v1.7.4/atom-amd64.deb
sudo dpkg --install atom-amd64.deb
sudo  apt-get -f install
```

3. Enable dart plugin
```
apm install dartlang
atom .
```

Wait for all packages to install, then quit Atom.

4. Install dartino
```
wget https://storage.googleapis.com/dartino-archive/channels/dev/release/latest/sdk/dartino-sdk-linux-x64-release.zip
unzip dartino-sdk-linux-x64-release.zip 
apm install dartino
```

5. Launch Atom Dartino
```
cd dartino-sdk/samples
atom .
```

6. Open a dartino program
   - Select /general/hello.dart in the file tree
   - Wait for Analyzing sources to complete

### Expected result

No analysis errors

### Actual result

Lots of errors and warnings

mit-mit, 2016-05-27:
CC @sgjesse 

We observed this issue on several developer machines at the IO codelab.

mit-mit, 2016-05-27:
Some thoughts: I think this is an architectural issue with the way our Atom plugins are structured: It looks like the dartlang plugin is both:

A) a concrete plugin for users writing Dart (for Web), and
B) a shared component providing base services to Flutter and Dartino

I think it would be better if we had:
1) A shared plugin for base services that is just services and no UI
2) A Dart (for Web) plugin that depends on 1) and has the UI for Dart (for Web)
3) A Dartino plugin that depends on 1), and has all the UI for Dartino (incl. what is in the current Dart plugin)
4) A Flutter plugin 

And then 2), 3), and 4) should be able to each run against separate Dart SDKs.

Is that possible?

danrubel, 2016-06-01:
Yes, this is possible, but the UI would suffer. If someone loaded more than one of the Dart, Dartino, and Flutter plugins, then the they would see multiple error views, multiple debug dialogs, etc. The analyzer from each plugin would try to analyze all projects containing dart files meaning the user would have multiple error views filled with different error messages.

That said, we are working towards better product separation. We have pulled much of the infrastructure out into a separate common [atom.dart repository](https://github.com/dart-atom/atom.dart). At the moment, we are constrained by our current build process from doing more. Once we can use DDC to compile our plugins, we will be able to build a richer API between plugins and we will refactor the plugins further.

I could not access the dart install from my Linux box. What version of the Dart SDK was installed when this problem occurred?

When I open the 0.4.0 samples, I see 2 warnings and 1 suggestion from analyzer... all of them correct.

mit-mit, 2016-06-02:
I just reproed again using the exact steps above on a clean ubuntu install.

I got this dart version:
```
Dart VM version: 1.16.1 (Thu May 26 01:07:47 2016) on "linux_x64"
```

And this dartino version:
```
0.4.0-dev.1.1
```

mit-mit, 2016-06-07:
@danrubel were you able to repro also?

mit-mit, 2016-06-29:
Bumping to 0.6.0

danrubel, 2016-06-30:
Do you have Flutter installed, or rather is `flutterRoot` defined in your Atom `config.cson` file?

I was able to repro, but only if `flutterRoot` is defined. This means that first time Dartino users who have not installed Flutter or the Atom Flutter plugin will not see this problem. That said, I'll continue to work on the problem.

mit-mit, 2016-07-20:
No, I don't. I only have what the instructions show above, which are run on a clean linux box.

I just tried again with dart 1.17.1-1 and atom 1.8.0 and the bug still repros.

I can share the VM with you, or I can I can share any files you need, just let me know.

mit-mit, 2016-07-20:
The full config.cson file from the VM:

```
user@vm:~/.atom$ cat config.cson 
"*":
  _dartlang:
    _errorsInitialized: true
    _firstRun: true
    _version: "0.6.32"
  "autocomplete-plus":
    autoActivationDelay: 500
  core:
    followSymlinks: false
  dartino:
    sdkPath: "/home/user/dartino-sdk"
  dartlang:
    sdkLocation: "/usr/lib/dart"
  "exception-reporting":
    userId: "2a255f10-4478-55b7-f178-a924c7e6f75c"
  linter:
    displayLinterInfo: false
    showErrorPanel: false
  welcome:
    showOnStartup: false
```