dart-archive/sdk #528: Setting breakpoint in the debugger crashes the debugger

sgjesse, 2016-06-03:
```
out/ReleaseX64/dartino debug out/DebugSTM/test.dart on tty /dev/ttyACM0
Using template settings file '/usr/local/prj/dart/github/dartino/sdk/.dartino-settings'
Attached to /dev/ttyACM0
Starting session. Type 'help' for a list of commands.

> b main
Closing tty-connection
Invalid argument(s)
#0      _TypedList._setUint64 (dart:typed_data-patch/typed_data.dart:885)
#1      _ByteDataView.setUint64 (dart:typed_data-patch/typed_data.dart:3434)
#2      CommandBuffer.addUint64 (package:dartino_compiler/src/shared_command_infrastructure.dart:146:10)
#3      MapAccess.internalAddTo (package:dartino_compiler/vm_commands.dart:491:11)
#4      VmCommand.addTo (package:dartino_compiler/vm_commands.dart:165:5)
#5      DartinoVmContext.sendCommand.<sendCommand_async_body> (package:dartino_compiler/vm_context.dart:229:13)
#6      Future.Future.microtask.<anonymous closure> (dart:async/future.dart:144)
#7      _rootRun (dart:async/zone.dart:895)
#8      _CustomZone.run (dart:async/zone.dart:790)
#9      _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:723)
#10     _microtaskLoop (dart:async/schedule_microtask.dart:41)
#11     _startMicrotaskLoop (dart:async/schedule_microtask.dart:50)
#12     _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#13     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:149)
```

sgjesse, 2016-06-03:
This happens using the `test.dart` we have when building.
```
ninja
ninja -C out/ReleaseX64
ninja -C out/ReleaseIA32
ninja -C out/DebugSTM disco_dartino_flash

cp platforms/stm/disco_dartino/src/test.dart out/DebugSTM
mv out/DebugSTM/snapshot out/DebugSTM/test.snapshot
mv out/DebugSTM/snapshot.info.json out/DebugSTM/test.snapshot.info.json

out/ReleaseX64/dartino debug out/DebugSTM/test.dart on tty /dev/ttyACM0
```
If I use `dartino flash` and `dartino debug` on the same file it does not happen
```
out/ReleaseX64/dartino flash out/DebugSTM/test.dart
out/ReleaseX64/dartino debug out/DebugSTM/test.dart on tty /dev/ttyACM0
```
Can it be caused by moving `test.dart` to a different directory from where it was build from?

sigurdm, 2016-06-03:
Yes - that sounds likely. The libraries in the info file are identified by absolute paths.
I guess we should try to make them relative to where the snapshot is located, so at least moving the two files together would work.