dart-archive/sdk #558: 0.5.0 cherry picks

mit-mit, 2016-06-30:
We should pick the following for the 0.5.0 stable release:

 - [ ] #557: unused import in volume.dart

sgjesse, 2016-07-04:
Version 0.5.0-dev.1.1 in https://github.com/dartino/sdk/commit/073336455adced76a6e1938c29313bf4fab58284

sgjesse, 2016-07-04:
Version 0.5.0-dev.1.2 in https://github.com/dartino/sdk/commit/75d4490dea0fc24c702c1eed1e3c8fd71dce1f55

sgjesse, 2016-07-04:
Version 0.5.0-dev.1.3 in https://github.com/dartino/sdk/commit/a8a9f2f3034b8fcf44bc40bd7d7a845b37b42cd5

sgjesse, 2016-07-04:
Version 0.5.0-dev.1.4 in https://github.com/dartino/sdk/commit/f3f682bcd7dbb63f79cda087a1a8ce1476c65d52

sgjesse, 2016-07-05:
Version 0.5.0-dev.1.5 in https://github.com/dartino/sdk/commit/5dfce8d8e48bc9b72c210e4e374d7fc87d486479

sgjesse, 2016-07-05:
Version 0.5.0-dev.1.5 was released.

closed 2016-07-05 by sgjesse