dart-archive/sdk #560: local debug causes NoSuchMethodError: method not found: 'listeners'

danrubel, 2016-06-30:
Starting debug session...
$ ..../dartino debug serve ...../main.dart
Using template settings file '...../dartino-sdk/internal/.dartino-settings'
localhost:62799
Connecting observatory to application on device...

The Dartino program is broken and has crashed.

The Dartino team would greatly appreciate if you would take a moment to
report this problem at https://github.com/dartino/sdk/issues/new

Please include the following information:

* the name and version of your operating system

* the Dartino SDK version (0.5.0-dev.1.0)

* the entire message you see here (including the full stack trace below)

The null object does not have a getter 'listeners'.

NoSuchMethodError: method not found: 'listeners'
Receiver: null
Arguments: []
#0      Object._noSuchMethod (dart:core-patch/object_patch.dart:42)
#1      Object.noSuchMethod (dart:core-patch/object_patch.dart:45)
#2      DebugConnection.serve.<serve_async_body> (package:dartino_compiler/src/debug_service_protocol.dart:485:15)
#3      Future.Future.microtask.<anonymous closure> (dart:async/future.dart:144)
#4      _rootRun (dart:async/zone.dart:895)
#5      _CustomZone.run (dart:async/zone.dart:790)
#6      _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:723)
#7      _microtaskLoop (dart:async/schedule_microtask.dart:41)
#8      _startMicrotaskLoop (dart:async/schedule_microtask.dart:50)
#9      _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#10     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:149)

danrubel, 2016-06-30:
It appears that [removing these lines](https://codereview.chromium.org/2109323002/patch/40001/50003) fixes the problem.

danrubel, 2016-06-30:
... but then I run into an Assertion failure: Node has not been computed .... so maybe that is not the right fix.

mit-mit, 2016-07-01:
CC @sigurdm

sigurdm, 2016-07-01:
I think the fix is mostly correct
I made a CL https://codereview.chromium.org/2115853002/ that only clears the vmContext after the invocation of the debugging server.
I don't see your assertion-failure - can you give more details?

commit e483c95a (2016-07-01): Don't clear the vmContext before it is used.

This closes #560.

R=karlklose@google.com

Review URL: https://codereview.chromium.org/2115853002 .
commit 01b72225 (2016-07-04): Don't clear the vmContext before it is used.

This closes #560.

R=karlklose@google.com

Review URL: https://codereview.chromium.org/2115853002 .
closed 2016-07-01 by sigurdm