dart-archive/sdk #582: NoSuchMethodError: method not found: 'getBeginToken' - when debugging

danrubel, 2016-07-11:
(test):~/work/git/dartino/weather/native$ /Users/danrubel/dartino-sdk/bin/dartino debug /Users/danrubel/work/git/dartino/weather/main.dart on tty /dev/tty.usbmodem1413
Using template settings file '/Users/danrubel/dartino-sdk/internal/.dartino-settings'
Attached to /dev/tty.usbmodem1413
Starting session. Type 'help' for a list of commands.

> lb
### no breakpoints
> bf onewire.dart 20
Closing tty-connection
The Dartino program is broken and has crashed.

The Dartino team would greatly appreciate if you would take a moment to
report this problem at https://github.com/dartino/sdk/issues/new

Please include the following information:

* Mac OSX 10.11.5

* the Dartino SDK version (0.5.0-edge.6c3fceed71e6aec16baadd77165b5c10b651e90a)

* the entire message you see here (including the full stack trace below)

The null object does not have a method 'getBeginToken'.

NoSuchMethodError: method not found: 'getBeginToken'
Receiver: null
Arguments: []
#0      Object._noSuchMethod (dart:core-patch/object_patch.dart:42)
#1      Object.noSuchMethod (dart:core-patch/object_patch.dart:45)
#2      FindFunctionAtPositionVisitor.containsPosition (package:dartino_compiler/src/find_position_visitor.dart:70:17)
#3      FindFunctionAtPositionVisitor.visitFunctionElement (package:dartino_compiler/src/find_position_visitor.dart:26:9)
#4      BaseElementVisitor.visitConstructorElement (package:compiler/src/elements/visitor.dart:160:12)
#5      SynthesizedConstructorElementX.accept (package:compiler/src/elements/modelx.dart:2406:20)
#6      FindFunctionAtPositionVisitor.visit (package:dartino_compiler/src/find_position_visitor.dart:65:32)
#7      LinkEntry.forEach (package:compiler/src/util/link_implementation.dart:120:8)
#8      ClassElementX.forEachLocalMember (package:compiler/src/elements/modelx.dart:2657:18)
#9      FindFunctionAtPositionVisitor.visitClassElement (package:dartino_compiler/src/find_position_visitor.dart:51:17)
#10     PartialClassElement.accept (package:compiler/src/parser/partial_elements.dart:421:20)
#11     FindFunctionAtPositionVisitor.visit (package:dartino_compiler/src/find_position_visitor.dart:65:32)
#12     LinkEntry.forEach (package:compiler/src/util/link_implementation.dart:120:8)
#13     CompilationUnitElementX.forEachLocalMember (package:compiler/src/elements/modelx.dart:715:18)
#14     FindFunctionAtPositionVisitor.visitCompilationUnitElement (package:dartino_compiler/src/find_position_visitor.dart:57:13)
#15     CompilationUnitElementX.accept (package:compiler/src/elements/modelx.dart:771:20)
#16     FindFunctionAtPositionVisitor.visit (package:dartino_compiler/src/find_position_visitor.dart:65:32)
#17     findFunctionAtPosition (package:dartino_compiler/src/find_position_visitor.dart:15:11)
#18     DartinoCompilerImplementation.debugInfoForPosition (package:dartino_compiler/src/dartino_compiler_implementation.dart:177:42)
#19     IncrementalCompiler.debugInfoForPosition (package:dartino_compiler/incremental/dartino_compiler_incremental.dart:294:22)
#20     DartinoVmContext.setFileBreakpointFromPosition.<setFileBreakpointFromPosition_async_body> (package:dartino_compiler/vm_context.dart:658:36)
#21     Future.Future.microtask.<anonymous closure> (dart:async/future.dart:144)
#22     _rootRun (dart:async/zone.dart:895)
#23     _CustomZone.run (dart:async/zone.dart:790)
#24     _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:723)
#25     _microtaskLoop (dart:async/schedule_microtask.dart:41)
#26     _startMicrotaskLoop (dart:async/schedule_microtask.dart:50)
#27     _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#28     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:149)

Session 'local' was terminated.
(test):~/work/git/dartino/weather/native$

sigurdm, 2016-07-12:
I'm not sure when this happens.
Can you send the source code?
I made a CL that will work around this issue when the element has no node.
https://codereview.chromium.org/2140053002/

sgjesse, 2016-08-01:
@danrubel Do you have a reproduction (using the `onewire.dart` source file)?