dart-archive/sdk #588: Debugging broken after latest Dart roll

jakobr-google, 2016-08-03:
I get the following error when I try to debug a target using eg. 'dartino debug samples/stm32f411re-nucleo/volume.dart on tty /dev/ttyACM0':

```
* the entire message you see here (including the full stack trace below)

Relative paths for dart extensions are not supported: 'src/serial_port'
Session 'local' was terminated.

```

Looks like path resolution for external libraries was changed recently on the Dart side, and now disallows relative paths. The third party serial_port library uses `import 'dart-ext:src/serial_port';`.

karlklose, 2016-08-03:
We need to change the location of the serial port library in the same way I
did with the other libraries in the roll.

closed 2016-08-03 by sgjesse

sgjesse, 2016-08-03:
Thanks for the info Karl. I will create a CL.

commit af933f98 (2016-08-03): Move the serial_port native code

The latest version of the Dart VM require native code for native
extensions to be in the same directory as the Dart file inporting it.

Closes #588

BUG= https://github.com/dartino/sdk/issues/588

R=jakobr@google.com
BUG=

Review URL: https://codereview.chromium.org/2206873004 .